kitspace/kitspace-v2 #5: Development Environment

SDBowen, 2019-12-22:
I got the Docker containers going easily with the readme instructions.

I did initially get the error below, due to not cloning the Gitea sub repository. Maybe mentioning that in the install process would be nice. 
```
Building gitea
ERROR: Cannot locate specified Dockerfile: Dockerfile
```
After completing the install and registering an account, I get this from `http://kitspace.test:3000/`

[image]

kasbah, 2019-12-22:
Ah, right. Nuke your own clone (`rm -rf gitea`). You need the [hacked version](https://github.com/kitspace/gitea/tree/kitspace-rebased) which you can get by running:

```
git submodule update --init
```

That needs to be added to the instructions.

SDBowen, 2019-12-22:
Yup, that did it.

[image]

commit 448b55b8 (2019-12-22): Add submodule instructions to readme

fixes #5
closed 2019-12-22 by kasbah