kitspace/kitspace-v2 #12: docker-compose up fails

SDBowen, 2020-06-07:
I get the below with `docker-compose up` and a clean install of the repo following the setup instructions.

```
frontend_1  | Compiled successfully.
frontend_1  | 
frontend_1  | Automatically optimizing pages...
frontend_1  | Warning: You have opted-out of Automatic Static Optimization due to `getInitialProps` in `pages/_app`. This does not opt-out pages with `getStaticProps`
frontend_1  | Read more: https://err.sh/next.js/opt-out-auto-static-optimization
frontend_1  | 
frontend_1  | 
frontend_1  | Page                                                           Size     First Load JS
frontend_1  | ┌ λ /                                                          1.18 kB         121 kB
frontend_1  | ├   /_app                                                      2.88 kB        76.9 kB
frontend_1  | ├ λ /1-click-bom                                               1.05 kB         114 kB
frontend_1  | ├ λ /404                                                       2.92 kB         116 kB
frontend_1  | ├ λ /bom-builder                                               2.96 kB         116 kB
frontend_1  | └ λ /projects/new                                              5.32 kB         125 kB
frontend_1  |     └ css/1a79ca7f41815bc658ec.css                             822 B
frontend_1  | + First Load JS shared by all                                  76.9 kB
frontend_1  |   ├ static/pages/_app.js                                       2.88 kB
frontend_1  |   ├ chunks/commons.afec7b.js                                   22.3 kB
frontend_1  |   ├ chunks/dd3b7df2062c54242cd1c1e50fe633e010029e80.207e29.js  4.01 kB
frontend_1  |   ├ chunks/framework.c61f0e.js                                 40.8 kB
frontend_1  |   ├ runtime/main.dc264a.js                                     6.27 kB
frontend_1  |   ├ runtime/webpack.1c5199.js                                  746 B
frontend_1  |   └ css/822514b03fd56455f2e7.css                               39.8 kB
frontend_1  | 
frontend_1  | λ  (Server)  server-side renders at runtime (uses getInitialProps or getServerSideProps)
frontend_1  | ○  (Static)  automatically rendered as static HTML (uses no initial props)
frontend_1  | ●  (SSG)     automatically generated as static HTML + JSON (uses getStaticProps)
frontend_1  | 
frontend_1  | > using build directory: /app/.next
frontend_1  | (node:111) ExperimentalWarning: The fs.promises API is experimental
frontend_1  |   copying "static build" directory
frontend_1  |   launching 3 workers
frontend_1  | Exporting (0/47)
frontend_1  |   copying "public" directory
frontend_1  | (node:124) ExperimentalWarning: The fs.promises API is experimental
frontend_1  | (node:130) ExperimentalWarning: The fs.promises API is experimental
frontend_1  | (node:132) ExperimentalWarning: The fs.promises API is experimental
frontend_1  | Exporting (1/47)
frontend_1  | Exporting (2/47)
gitea_1     | [Macaron] 2020-06-07 12:21:24: Started GET /api/v1/repos/search?sort=updated&order=desc for 172.22.0.2
frontend_1  | Exporting (3/47)
gitea_1     | [Macaron] 2020-06-07 12:21:24: Completed GET /api/v1/repos/search?sort=updated&order=desc 302 Found in 322.391µs
gitea_1     | [Macaron] 2020-06-07 12:21:24: Started GET /install for 172.22.0.2
gitea_1     | [Macaron] 2020-06-07 12:21:24: Started GET /api/v1/repos/search?sort=updated&order=desc for 172.22.0.2
gitea_1     | [Macaron] 2020-06-07 12:21:24: Completed GET /api/v1/repos/search?sort=updated&order=desc 302 Found in 252.878µs
gitea_1     | [Macaron] 2020-06-07 12:21:24: Started GET /install for 172.22.0.2
frontend_1  | Exporting (4/47)
frontend_1  | Exporting (5/47)
frontend_1  | Exporting (6/47)
frontend_1  | Exporting (7/47)
frontend_1  | Exporting (8/47)
frontend_1  | Exporting (9/47)
frontend_1  | Exporting (10/47)
frontend_1  | Exporting (11/47)
gitea_1     | [Macaron] 2020-06-07 12:21:24: Completed GET /install 200 OK in 122.447714ms
frontend_1  | 
frontend_1  | Error occurred prerendering page "/index". Read more: https://err.sh/next.js/prerender-error
frontend_1  | TypeError: Cannot read property 'map' of undefined
frontend_1  | Exporting (12/47)
frontend_1  | Exporting (13/47)
frontend_1  | Exporting (14/47)
gitea_1     | [Macaron] 2020-06-07 12:21:24: Completed GET /install 200 OK in 128.232474ms
frontend_1  | 
frontend_1  | Error occurred prerendering page "/". Read more: https://err.sh/next.js/prerender-error
frontend_1  | TypeError: Cannot read property 'map' of undefined
frontend_1  | Exporting (15/47)
frontend_1  | Exporting (16/47)
frontend_1  | Exporting (17/47)
frontend_1  | Exporting (18/47)
frontend_1  | Exporting (19/47)
frontend_1  | Exporting (20/47)
frontend_1  | Exporting (21/47)
frontend_1  | Exporting (22/47)
frontend_1  | Exporting (23/47)
frontend_1  | Exporting (24/47)
frontend_1  | Exporting (25/47)
frontend_1  | Exporting (26/47)
frontend_1  | Exporting (27/47)
frontend_1  | Exporting (28/47)
frontend_1  | Exporting (29/47)
frontend_1  | Exporting (30/47)
frontend_1  | Exporting (31/47)
frontend_1  | Exporting (32/47)
frontend_1  | Exporting (33/47)
frontend_1  | Exporting (34/47)
frontend_1  | Exporting (35/47)
frontend_1  | Exporting (36/47)
frontend_1  | Exporting (37/47)
frontend_1  | Exporting (38/47)
frontend_1  | Exporting (39/47)
frontend_1  | Exporting (40/47)
frontend_1  | Exporting (41/47)
frontend_1  | Exporting (42/47)
frontend_1  | Exporting (43/47)
frontend_1  | Exporting (44/47)
frontend_1  | Exporting (45/47)
frontend_1  | Exporting (46/47)
frontend_1  | Exporting (47/47)
frontend_1  | Error: Export encountered errors
frontend_1  |     at _default (/app/node_modules/next/dist/export/index.js:19:1164)
frontend_1  |     at process._tickCallback (internal/process/next_tick.js:68:7)
frontend_1  | error Command failed with exit code 1.
frontend_1  | info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
kitspace_frontend_1 exited with code 1

```

kasbah, 2020-06-07:
> TypeError: Cannot read property 'map' of undefined

I think getting the repos might be returning null:

https://github.com/kitspace/kitspace-using-gitea/blob/master/frontend/pages/index.js#L39-L52

You'll have to visit gitea.kitspace.test:3000 and set it up, maybe even import a repo.

kasbah, 2020-06-07:
Maybe the instructions should be (not sure):

```
docker-compose up gitea
```
Set up gitea on http://gitea.kitspace.test:3000 then:

```
docker-compose up
```

SDBowen, 2020-06-09:
Yeah, I just had to keep following the instructions.

> Maybe the instructions should be (not sure, haven't tested):

It would have to be:
```
docker-compose up nginx gitea
```

But maybe just make a note in step 4, "Frontend will fail until step 5 is complete" and include a final step of restarting docker compose?

kasbah, 2020-06-09:
I think bringing up the bits that are not going to fail first, then doing the rest is the way to go and instructions should be clearer on that.

commit c09e597e (2020-06-17): Add step to instructions in readme: run gitea first

closes #12
closed 2020-06-17 by kasbah