kitspace/kitspace-v2 #22: Integrate Custom Authentication Endpoints with Frontend

AbdulrhmnGhanem, 2020-08-26:
This is part of #6. The authentication flow should be more clear to solve the problems that originated from this [commit](https://github.com/kitspace/gitea/pull/3/commits/41d629e280a90f9b928a1611f21bdf1732a6589a).

>- The sign-in is broken, it authenticates correctly yet redirects to the Gitea homepage as if the user signed in from the Gitea? login page. A possible solution would be to just generate the token and send it to the frontend, let the frontend handle the reset of the sign-in flow.
>- Both reset password and forgot password are completely broken. They sign in the user. Probably this something to do with context.
>- How context will be passed from frontend to the backend.

AbdulrhmnGhanem, 2020-08-26:
Actually, after signing-in all whatever endpoint is called it signs the user in. Probably because of cookie although I see no cookies. :thinking:

kasbah, 2020-08-26:
You don't see any cookies? How are you looking for them? Gitea generally sets three cookies: `i_like_gitea`, `_csrf` and `lang`. I (can see them e.g. on https://staging.kitspace.org).

AbdulrhmnGhanem, 2020-08-26:
From the cookies tab in insomnia. I'll switch back to Postman.

AbdulrhmnGhanem, 2020-08-26:
> * A possible solution would be to just generate the token and send it to the frontend, let the frontend handle the reset of the sign-in flow.

> * How will context be passed from frontend to the backend? and the other way around?

Can you address those? Do you have a design in your mind?

kasbah, 2020-08-26:
I think it can work like this:

we POST `gitea.kitspace.test:3000/user/kitspace/sign_in` with JSON
- e.g. `{"user": "kaspar", "password": "********", "redirect_to": "/project/x"}`
- on success this responds with a re-direct but also sets the `i_like_gitea` session cookie
- the cookie is re-written by nginx to the apex domain (`.kitspace.test`)
https://github.com/kitspace/kitspace-using-gitea/blob/a4ade004639b6e96f7428befdbc6a61e42044c84/nginx/kitspace.template#L36
- it redirects to `kitspace.test:3000/project/x`

The user now has the session cookie and is authenticated on both `gitea.kitspace.test` and `kitspace.test`.  Do we need any more context than that?

I do feel it's getting a bit complicated by developing our own endpoints, that wasn't in the original plan. Let's discuss on a call later.

closed 2020-12-18 by AbdulrhmnGhanem