kitspace/kitspace-v2 #23: Figure out how to effeciently re-compile gitea in docker for development

kasbah, 2020-08-27:
It takes ages to re-compile at the moment.

kasbah, 2020-09-23:
One significant speed-up is to remove the npm and clean targets from the build in the Dockerfile. Not sure how best to integrate this as we may still want to re-run the JS stuff and clean occasionally. 

```diff
diff --git a/Dockerfile b/Dockerfile
index 9c678a60a..e13c0b081 100644
--- a/Dockerfile
+++ b/Dockerfile
@@ -20,7 +20,7 @@ WORKDIR ${GOPATH}/src/code.gitea.io/gitea
 
 #Checkout version if set
 RUN if [ -n "${GITEA_VERSION}" ]; then git checkout "${GITEA_VERSION}"; fi \
- && make clean-all build
+ && make backend
 
 FROM alpine:3.12
 LABEL maintainer="maintainers@gitea.io"
```

AbdulrhmnGhanem, 2020-12-19:
While deploying, I noticed the requirements for Gitea were pulled exceptionally quickly, perhaps AWS has all of them cached, this inspired me. Instead of relying on git submodule for Gitea, we can distribute it as a docker image this will make the versioning cleaner; there will be an explicit version number in the docker-compose file. The final image size for Gitea `146MB`. This isn't directly related to compilation time but I think it might add something.

kasbah, 2020-12-19:
Yes, there might be some caching/proxying but it's also just generally a much better network than a home network since it's a server.

I think there are two separate scenarios that we need to optimise for:

1. For development, you want to make changes to Gitea and spin it up within the dev environment  as quickly as possible. That's really what this issue is about, making that change->test cycle as quick as possible. Publishing the images for this doesn't seem like the right approach for development. 
2. For staging/production, yes, I think it would make sense to publish the image somewhere and then pull it in.

kasbah, 2020-12-19:
I've opened #34 for the container thing.

PR #36 (2020-12-20): Add a gitea dev docker container config
closes #23 

With this, `docker-compose up gitea` will run `make backend` before starting gitea, so our changes to gitea will be compiled in an effecient way.
closed 2020-12-20 by kasbah