kitspace/kitspace-v2 #25: Fix Layout shifts

AbdulrhmnGhanem, 2020-09-12:
Validation and success messages cause layout shifts when they get rendered.
I tried using one of the [examples](https://react.semantic-ui.com/collections/form/#shorthand-field-control-id) provided in Semantic-UI docs in a new page, not `/login/`, it gets rendered without any CSS!

AbdulrhmnGhanem, 2020-09-14:
The CSS problem was due to not importing the CSS for the component in `_app.js`

kasbah, 2020-09-14:
Ah yes, sorry I should have warned you about that. Maybe we need a comment for it somewhere. I considered just importing semantic.css, like we do on kitspace-v1 but it's about 600kb minified so would be good to just import what we need and this seems to be the only way to do that.

closed 2020-12-18 by AbdulrhmnGhanem