kitspace/kitspace-v2 #35: Husky-run not found

kasbah, 2020-12-20:
While it's a good idea to encourage running auto formatting I think adding git hooks via husky isn't a good fit for this project. The main issue is that not every folder is a javascript project, and husky might not be installed. I just ran into the issue while working on docker files: I couldn't commit because `husky-run` wasn't found. The solution was to remove the husky section from `frontend/package.json`.

AbdulrhmnGhanem, 2020-12-20:
If you installed husky globally the error should disappear.I usually use [pre-commit](https://pre-commit.com/#node) though it needs Python to get installed.

kasbah, 2020-12-22:
I just ran into it again. When I installed husky globally (`yarn global add husky`) it's still erroring:

```
$ yarn global add husky
$ which husky-run
/home/kaspar/.config/yarn/global/node_modules/.bin/husky-run
$ git commit -a
error Command "husky-run" not found.
```

I think it's causing more trouble than it's worth. If you want to install a pre-commit hook for yourself, then do that. Let's not bother with these tools that try and install it for other people. For enforcing formatting we should just get the CI to fail if it's not properly formatted.

AbdulrhmnGhanem, 2020-12-23:
Closed via  559bcf1

closed 2020-12-23 by AbdulrhmnGhanem