kitspace/kitspace-v2 #38: Set up a Github action to run the cypress tests

kasbah, 2020-12-22:


AbdulrhmnGhanem, 2020-12-23:
This will be a little tricky because of the Gitea setup setp. Probably will need a puppeteer script to do this step.

kasbah, 2020-12-23:
Ah yeah, there's probably a way to get Gitea to finish the DB setup and create an admin user but I haven't figured it out yet.

kasbah, 2020-12-23:
```
gitea admin create-user --username user01 --password password --email user01@example.com --admin
```

AbdulrhmnGhanem, 2021-01-22:
So for the CI, my plan is:
- Build all the docker images and tag them with the commit sha that triggered the workflow,
- run the tests.
- if all the tests pass tag the images `latest` and upload them to `ghcr`

Are OK with that?

kasbah, 2021-01-22:
Sounds like a plan.

AbdulrhmnGhanem, 2021-01-23:
> ```
> gitea admin create-user --username user01 --password password --email user01@example.com --admin
> ```

This line only creates the admin, the setup process involves more steps, for example, the database must be ready before issuing this command. Couldn't find such a command in the gitea command line [page](https://docs.gitea.io/en-us/command-line/). I guess a puppeteer script would be the easiest way to do the headless setup.

kasbah, 2021-01-23:
I mean, the database should be ready. That confirmation step via the web interface always seemed superflous to me. 

The above command I noticed when trying out this algola demo: https://github.com/camandel/agola-demo. I don't think it has any database setup script. Not sure what's different. Maybe we need to switch to sqlite for these tests?

AbdulrhmnGhanem, 2021-01-23:
I added the command as the last `RUN` step in the Dockerfile, actually, this exact command is outdated and it should be `gitea admin user create --username username --password password --email example@example.com --admin` but it complained about the database.

kasbah, 2021-01-23:
Maybe Gitea actually has to be running? In the Vagrant script it waits for it: https://github.com/camandel/agola-demo/blob/6d6ffbf31fb126f758e4fb03d76fbdb384a20a2f/Vagrantfile#L55

kasbah, 2021-01-23:
Dockerfile seems like the wrong place anyway, we don't want to bake this into our images.

AbdulrhmnGhanem, 2021-01-23:
Initially thought of `entrypoint.sh` though this won't run the first time only.

I wrote a puppeteer script that configures gitea will make a PR now so you can take look.

AbdulrhmnGhanem, 2021-01-24:
Running the puppeteer script staging turned out to be tedious, I don't think it will be a smooth process on GitHub actions. But I am not sure where to put the gitea initialization step.

AbdulrhmnGhanem, 2021-01-25:
Can we make a POST request to gitea? The installation page makes a request with the setup options, this it. A `curl` command in the `command` key in compose file should be sufficient. And then we wouldn't worry about whether this is a fresh installation or just running an existing setup.
Haven't tried it but probably will fail because of cookies.

kasbah, 2021-01-25:
Yes? If we need to send cookies then NodeJS with superagent might be easier.

AbdulrhmnGhanem, 2021-01-26:
Managed to get it to work with this script
```bash
# This script depends on `httpie`, install it from here `https://httpie.io/docs#linux`.
set -Eeuo pipefail

# Make a GET request to installation page, as if visiting '/index'
headers=$(http --header GET gitea.kitspace.test:3000)
# Store the cookies to make the request with
i_like_gitea=$(grep -oP '(?<=i_like_gitea=).*?(?=;)' <<<"$headers")
_csrf=$(grep -oP '(?<=_csrf=).*?(?=;)' <<<"$headers")

http --form --headers --check-status --ignore-stdin gitea.kitspace.test:3000 "Cookie:i_like_gitea="$i_like_gitea";_csrf="$_csrf"" \
	db_type='PostgreSQL' \
	db_host='postgres:5432' \
	db_user='gitea' \
	db_passwd='change_me' \
	db_name='gitea' \
	ssl_mode='disable' \
	charset='utf8' \
	db_path='/data/gitea/gitea.db' \
	app_name='Gitea: Git with a cup of tea' \
	repo_root_path='/data/git/repositories' \
	lfs_root_path='/data/git/lfs' \
	run_user='git' \
	domain='gitea.kitspace.test' \
	ssh_port='22' \
	http_port='3000' \
	app_url='http://gitea.kitspace.test:3000/' \
	log_root_path='/data/gitea/log' \
	enable_federated_avatar='on' \
	admin_name='__admin' \
	admin_passwd='123456' \
	admin_confirm_passwd='123456' \
	admin_email='admin@kitspace.com'

```

AbdulrhmnGhanem, 2021-01-28:
The good news, the tests are finally running in CI. The bad news, they are failing :laughing:

closed 2021-02-18 by kasbah