kitspace/kitspace-v2 #43: You can't upload files to mirrors

kasbah, 2020-12-23:
With https://github.com/kitspace/kitspace-v2/pull/30/commits/029054c2d034f594b3eab23b87e14e19b071f45a I made the imported projects mirrors. I didn't realise this but you can't actually upload files to mirrors. What I had hoped is that there would be two-way mirroring but it seems this isn't implemented yet (https://github.com/go-gitea/gitea/issues/2280). 

We do want to always pull in changes from the original repo so the options to resolve this are:

1. Don't allow file uploading to mirrors i.e. synced projects
2. Implement two way mirroring in Gitea

kasbah, 2020-12-23:
Thinking about it some more. even when two-way mirroring is implemented in Gitea, the user still needs to set up access to the remote repo (e.g. a Github personal access token). So we need to recognize that it's not implemented or set up yet and prevent people from trying to upload files to mirrors in the UI.

AbdulrhmnGhanem, 2020-12-24:
Should the project name and description be disabled too? 

Another possible solution would be giving an option to opt-out of mirroring and upload the files.

AbdulrhmnGhanem, 2020-12-25:
> the user still needs to set up access to the remote repo (e.g. a Github personal access token)

We can make a GitHub app that will ask the user for permission, the user won't need to go generate the access token and paste it in Kitspace.

commit 48f22b86 (2020-12-24): Prevent file upload for mirrored repos closes #43.
closed 2020-12-30 by kasbah