kitspace/kitspace-v2 #56: Nginx timeout when syncing big repos

AbdulrhmnGhanem, 2021-01-02:
Syncing a big repository e.g., [ulx3s](https://github.com/emard/ulx3s) takes a while which makes Nginx returns `504`. Currently, this is considered a failure, and an error message gets displayed as if syncing failed.

kasbah, 2021-01-02:
I think #58 is a bit of a workaround. I don't think it's reasonable of Gitea to wait to the import to complete before returning the response. It should probably be fixed upstream. I think we should also re-direct to a page that says "Importing..." , ignore the initial http response and then poll for completion. (I think that's how Gitea does it actually?)

AbdulrhmnGhanem, 2021-01-02:
I couldn't find a way to migrate from the GUI. Please, upload a screenshot.

kasbah, 2021-01-02:
Here, though it's not that important what Gitea itself does. It doesn't use the API so the response seems to be a 302 re-direct. 

[image]

AbdulrhmnGhanem, 2021-01-03:
> It doesn't use the API so the response seems to be a 302 re-direct.

Yes, It uses an internal endpoint `/{repo}/status` which returns `{errors: "", status: 1}` while the repo is still syncing  and ` {errors: "", status: 0}` when it's done. I looked into the source and it doesn't provide progress status.

> It doesn't use the API so the response seems to be a 302 re-direct.

This shold do the trick
```js
window.addEventListener("beforeunload", function() { debugger; }, false)
```

AbdulrhmnGhanem, 2021-01-03:
This problem is less severe in the staging server; the network is better than my crappy one :sweat_smile:

kasbah, 2021-01-03:
> This should do the trick
>> window.addEventListener("beforeunload", function() { debugger; }, false)

Huh, TIL about `debugger`, I used the network debug tab though which works fine too. 

> This problem is less severe in the staging server; the network is better than my crappy one 

Sure, we just have to find bigger git repos :)

AbdulrhmnGhanem, 2021-01-09:
> Yes, It uses an internal endpoint `/{repo}/status` which returns `{errors: "", status: 1}` while the repo is still syncing and ` {errors: "", status: 0}` when it's done. I looked into the source and it doesn't provide progress status.

This won't work out of the box for repositories migrated by the API:
- The `/status` endpoint relies on having a `migarationTask` to respond with the right status, see [gitea/.../repo.go](https://github.com/kitspace/gitea/blob/6bf78d2b576a5caa77a670db9bbf2572d0272f25/routers/repo/repo.go#L541)
-  Repos migrated by the API follows a different approach; it doesn't create a `migrationTask` at all:
     1.  It creates a repository, [here](https://github.com/kitspace/gitea/blob/2c5b3c72e8ebdf94ac764267fa78e4a0c5e93dc6/routers/api/v1/repo/migrate.go#L147).
     2.  After some validation it calls `migrations.MigrateRepository` directly, [here](https://github.com/kitspace/gitea/blob/2c5b3c72e8ebdf94ac764267fa78e4a0c5e93dc6/routers/api/v1/repo/migrate.go#L186).
 - Repos migrate by the UI
     1. same as API
     2. It calls `task.MigrateRepository`, [here](https://github.com/kitspace/gitea/blob/6bf78d2b576a5caa77a670db9bbf2572d0272f25/routers/repo/repo.go#L391).

Why is it done in two different approaches? I have no idea!

PR #58 (2021-03-25): Use polling when syncing repos
closes #56