kitspace/kitspace-v2 #60: Gitea frontend is broken

AbdulrhmnGhanem, 2021-01-05:
After using the new dockerfile configuration on the #57 branch, the frontend is broken(CSS files aren't found) when reverted back it worked fine

kasbah, 2021-01-05:
By dockerfile you mean `docker-compose.yml`, specifically the /etc/hosts mounting?

AbdulrhmnGhanem, 2021-01-05:
Yes, the compose file, but I rolled back using using the dev version of
gitea to using the original file, not the /etc/hosts

kasbah, 2021-01-05:
What does one have to do with the other?

AbdulrhmnGhanem, 2021-01-06:
See https://github.com/kitspace/kitspace-v2/pull/57/commits/2292429e061c29938d7a4a982e58ce50e011e3fe, the commenting those to lines fixed the Gitea frontend

AbdulrhmnGhanem, 2021-02-04:
@kasbah, try running `docker images prune -a` not `system prune`. This may reproduce the error for you.

AbdulrhmnGhanem, 2021-03-17:
Using `make build` instead of `make backend` in the command fixes the issue. The first time the container runs it will generate the assets. For subsequent builds, this step won't run `bindata for public already up-to-data`.

AbdulrhmnGhanem, 2021-03-17:
Closed by 14eb006, and https://github.com/kitspace/gitea/commit/b122f52663c4f575e4d228df1c8b901e1395b1b3

closed 2021-03-17 by kasbah