kitspace/kitspace-v2 #77: Bit worried about the keys used for SWR

kasbah, 2021-03-26:
This one seems redundantly embeds the params into the endpoint and additinally an object (where order and equality isn't guaranteed). 

https://github.com/kitspace/kitspace-v2/blob/2211fbf6908014f2209653ac085df005091f107f/frontend/src/hooks/Gitea.js#L86-L89

This one has a ternary, either using e.g.  `${giteaApiUrl}/users/null/repos` or most likely the bare `username`, as a key for the repos, which doesn't make a lot of sense to me.  

https://github.com/kitspace/kitspace-v2/blob/2211fbf6908014f2209653ac085df005091f107f/frontend/src/hooks/Gitea.js#L108-L111

kasbah, 2021-03-26:
I don't understand this one, even after looking at the docs: https://github.com/kitspace/kitspace-v2/blob/2211fbf6908014f2209653ac085df005091f107f/frontend/src/hooks/Gitea.js#L135

Why have a function that just returns the pre-computed `endpoint`?

kasbah, 2021-03-26:
Understanding 1. a bit better but I do think using an object there will cause trouble. Found this in the SWR docs:

> 
> ```js
> // Don’t do this! Deps will be changed on every render.
> useSWR(['/api/user', { id }], query)
> 
> // Instead, you should only pass “stable” values.
> useSWR(['/api/user', id], (url, id) => query(url, { id }))
> ```

https://swr.vercel.app/docs/arguments#passing-objects

AbdulrhmnGhanem, 2021-03-26:
> 1. This one redundantly embeds the params into the endpoint and additionally into an object (where order and equality aren't guaranteed).

Yes, after updating the endpoint I forgot to delete the params, this object is no longer needed.

> 2. This one has a ternary, either using e.g.  `${giteaApiUrl}/users/null/repos` or most likely the bare `username`, as a key for the repos, which doesn't make a lot of sense to me.

It will use this as a key `username ? endpoint : null`:
* if the `username` is `undefined` then it will evaluate to `null` which will prevent SWR from performing the request.
* Otherwise it will use the `endpoint` as a key, which contains the full repo name(username, and repo name)
Anyways I added this condition to get around a limitation that no longer exists; will remove it.

> I don't understand this one, even after looking at the docs:
> 
> https://github.com/kitspace/kitspace-v2/blob/2211fbf6908014f2209653ac085df005091f107f/frontend/src/hooks/Gitea.js#L131-L135
> 
> Why have a function that just returns the pre-computed `endpoint`?

This a mistake from me; forgot to update the comment and remove the function, it used to be like that 
```js
  // Dependent data if: swr won't fetch if `repoDetails.default_branch` returns a falsy value.
  // See https://swr.vercel.app/docs/conditional-fetching
  const { data, error, mutate } = useSWR(
    () => `${giteaApiUrl}/repos/${repo}/contents?ref=` + repoDetails.default_branch,
    fetcher,
    swrOpts,
  )
``` 
Modified how it worked in https://github.com/kitspace/kitspace-v2/commit/2c86ba944b03de6bb2562b699b682bc0470ee579 and left this dead comment.

kasbah, 2021-03-27:
Ok, got it. Read 2. wrong. Rest are small mistakes.

kasbah, 2021-07-18:
Looks like this one was never fixed: https://github.com/kitspace/kitspace-v2/blob/1ce88a4ff29a9dbd035e2d2391cd6c627c220fbf/frontend/src/hooks/Gitea.js#L80 

May be solved by doing search differently for #78

AbdulrhmnGhanem, 2021-07-29:
Fixed in #175 
https://github.com/kitspace/kitspace-v2/blob/65290b345fc1c8d494d2e6f95b5fab2a9f315892/frontend/src/hooks/Gitea.js#L76-L84

closed 2021-08-19 by kasbah