kitspace/kitspace-v2 #78: / route is changed to `/search`

kasbah, 2021-03-26:
The spec from #69 was

> Proxy to to /search but still appears as /. 

But it now appears as /search

AbdulrhmnGhanem, 2021-03-31:
I couldn't do that without page reload. Shallow routing is not supported between different pages.

AbdulrhmnGhanem, 2021-03-31:
I fixed it, but when some page redirects to `/` it gets displayed as `/search`. Duplicating the structure of `/search` in `/` will be problematic to maintain.

kasbah, 2021-03-31:
Hmm, that's not so great either. I think we just have to call it `/` i.e. index.js for now for simplicity. 

The reason for all of this is to avoid breaking links if we want to change `/` to something else (e.g. a dashboard) in the future. I guess we'd have to figure out how to redirect `/q=xyz` links to `/search?q=xyz` correctly in the future then.

kasbah, 2021-07-18:
I've just been looking more closely at this . I don't think Gitea will be able to cope with search as you type for very many projects (e.g. 300ms request time on try.gitea.io), hence:

1. Turn the search input into a proper form with a submit button 
2. Go to `/search?q=` when you submit 

That should simplify things overall.

PR #175 (2021-08-19): Fix redirecting to `/search` 
Fixes #78
closed 2021-08-19 by kasbah