kitspace/kitspace-v2 #79: `csrf` can be empty string

kasbah, 2021-03-26:
I don't think it makes sense that we send out requests that need the token before retrieving it, we shouldn't allow the token to be empty. Having some trouble with this while trying to make a new page.

AbdulrhmnGhanem, 2021-03-26:
I am not sure I understood this right!

* If you need the user to be logged in add `reqSignIn` prop to the page component like this[ `<Page reqSignIn>{...}</Page>`](https://github.com/kitspace/kitspace-v2/blob/2211fbf6908014f2209653ac085df005091f107f/frontend/src/pages/projects/new.js#L34).
* If it doesn't require the user to be logged in chances are `_csrf` isn't needed to perform the request(which is probably a GET request)

kasbah, 2021-03-28:
You might be right. It just doesn't _feel_ right that the csrf can be an empty string at times.  I was also surprised to find out that Gitea doesn't check the csrf on the login/register forms. ([Maybe it should](https://stackoverflow.com/a/15350123), though it's fairly common no to it seems.)

What I ended up doing is [checking for the csrf is not empty](https://github.com/kitspace/kitspace-v2/pull/80/files#diff-f6e3ede6e31f3955088c7ee3ebbfe67a931c262abaf7a84ea89367b210b98322R21) with SWR conditional fetching. I will have to use `reqSignIn` first before I can make further suggestions. (Btw. can we not rename it to `requireSignIn`? Reads much better I think, especially since `req` often means `request`).

AbdulrhmnGhanem, 2021-03-29:
> I was also surprised to find out that Gitea doesn't check the csrf on the login/register forms.

Actually, it does. The `_csrf` is sent with each form submission but it is guaranteed the `_csrf` will be by the time the form loads. 
I faced your problem while trying to use the of `csrf` from the `AuthContext` on page load, it takes a few milliseconds to propagate.
 
> (Btw. can we not rename it to `requireSignIn`? Reads much better I think, especially since `req` often means `request`).

I used the same convention as gitea, Though changing it won't be a problem.

kasbah, 2021-03-29:
The _csrf isn't verified by Gitea on their sign-up form or our endpoint though. I tried removing it and everything still seems to work. I wonder if I should report it.

kasbah, 2021-03-31:
Going to close this, seems this was mostly about me understanding how to get the csrf token now. It is an empty string for about 3ms at the start but hopefully that should never be a problem.

closed 2021-03-31 by kasbah