kitspace/kitspace-v2 #84: Heavy use of sed in publish-docker-images

kasbah, 2021-03-28:
Bit worried about these complicated `sed` calls in publish-docker-images. This one is especially bad:

https://github.com/kitspace/kitspace-v2/blob/04b069f429d4b5851b1e5933ed73aaf932cad8ed/.github/workflows/publish-docker-images.yml#L125-L138

Can we not get this done with environmental variables in the docker-compose.deploy.yml instead?

kasbah, 2021-03-28:
Additionally, would be good if we could re-use `.env.example` for all of these:

https://github.com/kitspace/kitspace-v2/blob/04b069f429d4b5851b1e5933ed73aaf932cad8ed/.github/workflows/publish-docker-images.yml#L2-L15

AbdulrhmnGhanem, 2021-03-29:
The problem is that the base image tag is changed to match the current commit, I couldn't think of any way to do that. Is even possible to read the name of the base image from environment variables. Port forwarding can be solved by another compose file but I don't think the rest can be.

> Additionally, would be great if we could re-use `.env.example` for all of these:
> 
> https://github.com/kitspace/kitspace-v2/blob/04b069f429d4b5851b1e5933ed73aaf932cad8ed/.github/workflows/publish-docker-images.yml#L2-L15
> 
> Not sure if there is an easy way to do it though.

We can `source .env.example` in a separate action before running this action as explained [here](https://docs.github.com/en/actions/reference/workflow-commands-for-github-actions#setting-an-environment-variable). But I opted for inlining it to make the action self-contained.

kasbah, 2021-03-29:
I think it would be better to source the `.env.example`. That way we get a reminder to update it from broken builds. Otherwise we need to remember to update the .example _and_ the workflow .yml when we change anything about the env variables. 

Can we not do something like this?
```yml

    image: 'ghcr.io/kitspace/frontend-v2:${FRONTEND_IMAGE_VERSION:-latest}'

```
And then:
```
FRONTEND_IMAGE_VERSION=${{ steps.sha.outputs.sha_short }} docker-compose -f docker-compose.deploy.yml up
```

Or probably best to set all of them in  `env:` section in the yaml.

commit 36c88e0e (2021-04-01): Make docker-compose.deploy.yml extend default again

Gets rid of nasty sed calls too (closes #84)
commit 1864d950 (2021-04-02): Make docker-compose.deploy.yml extend default again

Gets rid of nasty sed calls too (closes #84)
closed 2021-04-02 by kasbah