kitspace/kitspace-v2 #85: Remove initOps from fetcher

kasbah, 2021-03-28:
Including `initOps` confused me, the name is not clear ("initial" but then I would actually pass them to `useSWR(x, initOps)`?) Also OPS generally means "operations" and OPTS means options. Also the documentation is wrong:

https://github.com/kitspace/kitspace-v2/blob/2211fbf6908014f2209653ac085df005091f107f/frontend/src/hooks/Gitea.js#L7

Anyway, that's a lot of words about one function argument. I think we should remove it.

kasbah, 2021-03-28:
I guess it could be useful for `POST` requests sometime. Let's just call it something else: `fetchOpts` seems pretty clear. 

Also, I think this header is technically wrong:

https://github.com/kitspace/kitspace-v2/blob/2211fbf6908014f2209653ac085df005091f107f/frontend/src/hooks/Gitea.js#L29

It says you are sending JSON but then your are doing GET requests (and sending URL query params) so it's not really relevant. I think you want `{ 'Accept': 'application/json' } `?

AbdulrhmnGhanem, 2021-03-29:
> Including `initOps` confused me, the name is not clear ("initial" but then I would actually pass them to `useSWR(x, initOps)`?) Also OPS generally means "operations" and OPTS means "options". 

Yes, a `t` is missing. Though it explicitly says what it's doing for the docstring for the fetcher, also, the function doesn't call `useSwr` in the first place. 
https://github.com/kitspace/kitspace-v2/blob/2211fbf6908014f2209653ac085df005091f107f/frontend/src/hooks/Gitea.js#L34

> Also the documentation is wrong:
> https://github.com/kitspace/kitspace-v2/blob/2211fbf6908014f2209653ac085df005091f107f/frontend/src/hooks/Gitea.js#L7
> 
> Anyway, that's a lot of words about one function argument. ~I think we should remove it.~

This is the documentation of `useSwr`, not `fetcher`
https://github.com/kitspace/kitspace-v2/blob/2211fbf6908014f2209653ac085df005091f107f/frontend/src/hooks/Gitea.js#L5

> I guess it could be useful for `POST` requests sometime. Let's just call it something else: `fetchOptions` seems much clearer.

Why would we cache the response for a post request?

> Also, I think this header is technically wrong:
>
> https://github.com/kitspace/kitspace-v2/blob/2211fbf6908014f2209653ac085df005091f107f/frontend/src/hooks/Gitea.js#L29
>
> It says you are sending JSON [...]

Yes, indeed. I didn't know that.

kasbah, 2021-03-29:
> Though it explicitly says what it's doing for the docstring for the fetcher, also, the function doesn't call useSwr in the first place.

Kind of, the thing that isn't clear at all, as an outsider, is that you are meant to use it like this:

```js
userSwr([endpoint, fetchOptions], fetcher)
```

I think the docs and name should reflect that. 

> Why would we cache the response for a post request?

Touché. What is it useful for then? The only other thing I can think of is for adding headers e.g. for tidying (?) up the csrf:

```js
userSwr([endpoint, {headers: {...headers, 'X-Csrf-Token': csrf}}], fetcher)
```

kasbah, 2021-03-29:
But then, we're not even meant to use objects inside the SWR key? So I think it's not a good option.

AbdulrhmnGhanem, 2021-03-29:
I didn't get what you meant, let's discuss this next session!

kasbah, 2021-03-29:
You don't get anything of what I said!? Would be better if you clarified what parts you don't understand. It's not really worth discussing as much as we have been so I'm not sure doing it in person as well is helpful. Just needs to be cleaned up a little.

AbdulrhmnGhanem, 2021-03-29:
From what I understood I will rewrite it like this
```js
/**
 *
 * @param url
 * @returns {Promise<Promise<any>>}
 */
const fetcher = url =>
  fetch(url, {
    method: 'GET',
    mode: 'cors',
    headers:  { 'Accept': 'application/json' },
  }).then(r => r.json())

```

We don't make use of `csrf` here so the `'X-Csrf-Token': csrf` header isn't useful. I don't see we will need any other options for `fetch`; make fetcher isn't needed.

kasbah, 2021-03-29:
Yep, sounds good. Sorry to use so many words to describe something so simple. :sweat_smile:

PR #93 (2021-04-01): Clean gitea hooks
closes #85 #77
commit 1a7308d4 (2021-04-01): Clean gitea hooks (#93)

* Improve `fetcher`(closes #85).

- Remove options,
- use correct headers.

* Make populate condition defaults to true.

* Delete `useSearchRepo` hook.

- Relying on the data returned by next from ssr is totally sufficient;
no need for a hook, next handle updating the data for us.

* Delete dead code in `hooks/Gitea.js`.

* Increase timeout in `updateProject.spec` before hook.

* Retore `useSearchRepos` in `hooks/Gitea.js`

- Use `AllRepos` relies on it, search
results in `/` no longer relies on it, it's not dead.
closed 2021-04-01 by kasbah