kitspace/kitspace-v2 #100: The processor stuck at `addDir` when processing the first added repo.

AbdulrhmnGhanem, 2021-04-10:
Sometimes the after migrating the first repo to gitea, the processor gets stuck at the `addDir` step. restarting the container fixes the problem.

kasbah, 2021-04-13:
I managed to reproduce this, but only once. If it happens again can you try and capture the full log output from the processor please?

AbdulrhmnGhanem, 2021-04-15:
I ran `sudo rm -rf volumes` then the normal setup steps.
```sh
processor_1  | yarn install && yarn dev
processor_1  | yarn install v1.22.5
processor_1  | [1/4] Resolving packages...
processor_1  | success Already up-to-date.
processor_1  | Done in 0.54s.
processor_1  | yarn run v1.22.5
processor_1  | $ nodemon src/server.js --watch src/
processor_1  | [nodemon] 2.0.6
processor_1  | [nodemon] to restart at any time, enter `rs`
processor_1  | [nodemon] watching path(s): src/**/*
processor_1  | [nodemon] watching extensions: js,mjs,json
processor_1  | [nodemon] starting `node src/server.js`
processor_1  | processor listening on http://localhost:5000
processor_1  | addDir /repositories/__admin/ch330_hardware.git
processor_1  | cloned into /data/checkout/__admin/ch330_hardware
processor_1  | (node:62) UnhandledPromiseRejectionWarning: Error: Command failed: cd '/data/checkout/__admin/ch330_hardware' && git rev-parse HEAD
processor_1  | fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.
processor_1  | Use '--' to separate paths from revisions, like this:
processor_1  | 'git <command> [<revision>...] -- [<file>...]'
processor_1  | 
processor_1  |     at ChildProcess.exithandler (child_process.js:308:12)
processor_1  |     at ChildProcess.emit (events.js:315:20)
processor_1  |     at maybeClose (internal/child_process.js:1048:16)
processor_1  |     at Socket.<anonymous> (internal/child_process.js:439:11)
processor_1  |     at Socket.emit (events.js:315:20)
processor_1  |     at Pipe.<anonymous> (net.js:673:12)
processor_1  | (Use `node --trace-warnings ...` to show where the warning was created)
processor_1  | (node:62) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
processor_1  | (node:62) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
processor_1  | addDir /repositories/__admin/ch330_hardware.git
processor_1  | addDir /repositories/__admin/ch330_hardware.git
^CGracefully stopping... (press Ctrl+C again to force)
Stopping kitspace_processor_1 ... done
☁  kitspace-v2 [project-page] ⚡  docker-compose up processor
Starting kitspace_processor_1
Attaching to kitspace_processor_1
processor_1  | yarn install && yarn dev
processor_1  | yarn install v1.22.5
processor_1  | [1/4] Resolving packages...
processor_1  | success Already up-to-date.
processor_1  | Done in 0.54s.
processor_1  | yarn run v1.22.5
processor_1  | $ nodemon src/server.js --watch src/
processor_1  | [nodemon] 2.0.6
processor_1  | [nodemon] to restart at any time, enter `rs`
processor_1  | [nodemon] watching path(s): src/**/*
processor_1  | [nodemon] watching extensions: js,mjs,json
processor_1  | [nodemon] starting `node src/server.js`
processor_1  | processor listening on http://localhost:5000
processor_1  | addDir /repositories/__admin/ch330_hardware.git
processor_1  | in_progress __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/ch330_hardware-7a79f83-gerbers.zip
processor_1  | in_progress __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/images/bottom.svg
processor_1  | in_progress __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/zip-info.json
processor_1  | in_progress __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/images/top.svg
processor_1  | in_progress __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/images/top.png
processor_1  | in_progress __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/images/top-large.png
processor_1  | in_progress __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/images/top-meta.png
processor_1  | in_progress __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/images/top-with-background.png
processor_1  | in_progress __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/1-click-BOM.tsv
processor_1  | in_progress __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/info.json
processor_1  | in_progress __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/interactive_bom.json
processor_1  | done __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/images/bottom.svg
processor_1  | done __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/zip-info.json
processor_1  | done __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/images/top.svg
processor_1  | done __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/ch330_hardware-7a79f83-gerbers.zip
processor_1  | done __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/info.json
processor_1  | done __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/1-click-BOM.tsv
processor_1  | done __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/interactive_bom.json
processor_1  | done __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/images/top.png
processor_1  | done __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/images/top-large.png
processor_1  | done __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/images/top-meta.png
processor_1  | done __admin/ch330_hardware/7a79f8372914275077f0afd97e32b04bd02032a6/images/top-with-background.png
```

kasbah, 2021-04-16:
Perfect, thanks that git error is the one I was trying to catch.

> sudo rm -rf volumes

By the way, the volumes on master are no longer on disk in that dir. So when you merge/rebase, an easy way to remove them all is `docker-compose down -v`.

AbdulrhmnGhanem, 2021-04-23:
The [Pyr0-Piezo](https://kitspace.org/boards/github.com/pyr0ball/pyr0piezo/pyr0-piezo%20z-probe%20business%20card%20diy%20module/) breaks the processor. 
```sh
yarn run v1.22.5
$ node src/server.js
processor listening on http://localhost:5000
(node:28) UnhandledPromiseRejectionWarning: Error: Command failed: cd '/data/checkout/tester/pyr0piezo' && git rev-parse HEAD
fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'

    at ChildProcess.exithandler (child_process.js:308:12)
    at ChildProcess.emit (events.js:315:20)
    at maybeClose (internal/child_process.js:1048:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:28) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:28) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
yarn run v1.22.5
$ node src/server.js
processor listening on http://localhost:5000
repo without any branches /data/checkout/tester/pyr0piezo
```

Now the processor is too quiet that I don't know what is going on :laughing:

kasbah, 2021-04-23:
You can set `LOG_LEVEL=debug` via an env. That project is using `multi` which isn't supported at all yet.

AbdulrhmnGhanem, 2021-04-23:
> That project is using `multi` in the kitspace.yaml which isn't supported at all yet, though I'm not sure the error is related.

Interestingly, The [thumbnail](https://abdo-dev.staging.kitspace.dev/tester/) is generated for it.

kasbah, 2021-04-23:
Well, it's not that interesting... it just searches for stuff when it can't find the right fields in kitspace.yaml.

AbdulrhmnGhanem, 2021-07-14:
The processor keeps producing this error.

```
Error processing '/gitea-data/git/repositories/sally/diy_particle_detector.git': Error: Command failed: cd '/data/checkout/sally/diy_particle_detector' && git rev-parse HEAD
processor_1  | fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.
processor_1  | Use '--' to separate paths from revisions, like this:
processor_1  | 'git <command> [<revision>...] -- [<file>...]'
processor_1  | 
processor_1  |     at ChildProcess.exithandler (child_process.js:319:12)
processor_1  |     at ChildProcess.emit (events.js:375:28)
processor_1  |     at maybeClose (internal/child_process.js:1055:16)
processor_1  |     at Socket.<anonymous> (internal/child_process.js:441:11)
processor_1  |     at Socket.emit (events.js:375:28)
processor_1  |     at Pipe.<anonymous> (net.js:675:12) {
processor_1  |   killed: false,
processor_1  |   code: 128,
processor_1  |   signal: null,
processor_1  |   cmd: "cd '/data/checkout/sally/diy_particle_detector' && git rev-parse HEAD",
processor_1  |   stdout: 'HEAD\n',
processor_1  |   stderr: "fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.\n" +
processor_1  |     "Use '--' to separate paths from revisions, like this:\n" +
processor_1  |     "'git <command> [<revision>...] -- [<file>...]'\n"
processor_1  | }
```

kasbah, 2021-07-14:
That seems unrelated to the previous issues. Can you please open separate issues for different problems with the processor.

kasbah, 2021-07-14:
Actually sorry, that does seem to be the same as the original error.

AbdulrhmnGhanem, 2021-07-14:
The easiest way to reproduce is to run the `multiPart` tests.

AbdulrhmnGhanem, 2021-07-21:
Also, the processor status endpoint returns `404` when this happens, it should return
```
{
  "status": "failed",
  "error": "..."
}
```

AbdulrhmnGhanem, 2021-08-05:
I think this happens because the processing event fires before Gitea finishes the migration.
If it turned this was the cause, the debouncing should be replaced with a queue, processing tasks should be added to the queue when the migration is done.

PR #234 (2022-01-26): Fix premature processing for gitea repos
Fixes #100
## 
- When migrating a repo [1]: Gitea creates an empty repo then the migration
  starts.
- When uploading user files [2]: we create an empty repo,
  upload the files, finally commit the files to the repo.
- When the processor gets notified by a file-system change when the
  repository is created in both [1], and [2] the repo is still empty and
  the `git checkout` command fails.
- The fix:
    - only process repos if it isn't empty,
    - only process synced repos when the migration task is done.
 ##
* The processor check gitea database for  [... cut]
closed 2022-01-26 by kasbah