kitspace/kitspace-v2 #125: Fix 1-click-bom extension integration.

AbdulrhmnGhanem, 2021-05-29:


kasbah, 2021-06-18:
Port :3000 isn't allowed in Chrome for some reason so I had to upload a v1.7.5 version with just the `kitspace.test` domain (i.e. port 80). `kitspace.dev` is still enabled though and it's been released.

AbdulrhmnGhanem, 2022-11-15:
I will add a `/1-click-BOM.tsv` route to the project page and make it redirect the actual asset URL. This will let us control the asset URL without modifying the extension, just change the redirect URL.

@kasbah, what do you think of this?

kasbah, 2022-11-15:
ok

kasbah, 2023-05-28:
I've added the `/1-click-BOM.tsv` route and fixed up the extension integration with #587. I've also updated 1clickBOM and pushed a new version to Google and Mozilla extension stores. Once #587 is merged in and new extensions are approved by the stores we can close this issue.

PR #587 (2023-06-28): Connect up 1-click-BOM extension
fixes #125
closed 2023-06-28 by AbdulrhmnGhanem