kitspace/kitspace-v2 #129: Issues with server-side vs client-side page loads

kasbah, 2021-05-30:
On some page transitions the user isn't loaded properly if you change page client-side. Especially noticeable around the user's logged-in state and project ownership (i.e. the file selection buttons don't show up unless you reload). Might this be because of heavy use of `getServerSideProps`. Would `getInitialProps` be a better fit to what we are doing? What exactly is the difference?

kasbah, 2021-05-30:
Found this discussion: https://github.com/vercel/next.js/discussions/11211

kasbah, 2021-06-03:
So @AbdulrhmnGhanem, you linked me [this change](https://github.com/kitspace/kitspace-v2/commit/9ee25a85f3491f51aff39bf9f006c828c007f697#diff-1c9b90fc0806fbbdb564cf564a3b7a49ef8cf615dc132f8bc8f68de74cfdecec) which solves this in a way. It does feel like a work-around though. Can you explain why this is needed. From discussions above it shouldn't be needed. 

I also came across [this issue](https://github.com/vercel/next.js/issues/13910) where a lot of people are complaining about extra fetching needed for `getServerSideProps`  and NextJS creators mention `getInitialProps` won't be removed any time soon. Maybe `getInitialProps` is a better solution for us?

AbdulrhmnGhanem, 2021-06-07:
To check the user, who is making the request, not the project owner,  `req?.session?.user?.username` is used. For client-side navigation the value of `req.session` is `null`, this only happens in production as in development all navigations rebuild the page.

> Maybe getInitialProps is a better solution for us?

The [docs](https://nextjs.org/docs/api-reference/data-fetching/getInitialProps) recommend using `getStaticProps`, this the only reason that made me use it. In theory, renaming `getStaticProps` to `getInitialProps` will be the only required change to start using it.

kasbah, 2021-06-07:
That makes sense. We need to have a good way to persist the session that doesn't depend on `req.session` being present then. 

The question about switching from `getStaticProps` to `getInitialProps` seems to be mainly a performance trade-off, initial bundle size vs page-transition requests, so let's hold off on that.

AbdulrhmnGhanem, 2021-09-19:
In 634c77b, I tried `getServerSideProps` for multi-part projects and `getInitialProps` for normal projects. And there's no difference; the `req` object isn't available with client-side navigation, it's mentioned in the documentation for `getInitialProps` but not with `getStaticProps` so I had to test it.

To unify the logic I think we should do anything that relies on `session` in the frontend only.

AbdulrhmnGhanem, 2021-09-19:
Also, take a look at [next-express](https://github.com/shdnx/next-express). Will it work given how the token is passed from gitea now?

kasbah, 2021-09-20:
`req` not being there is expected to me. That's why I attached the session to `window` in my initial implementation (maybe you are still doing something akin to this, I'm not sure). To me there isn't really a problem here, it just needs to be handled correctly for both cases, server and client side, `req` being there and not. (Don't have time to look into next-express right now but at a quick glance it's not going to solve any of this for us.)

> To unify the logic I think we should do anything that relies on session on the client-side only.

That doesn't make sense to me. It's there as `req.session` on the server-side, no?

AbdulrhmnGhanem, 2021-09-20:
This means I will duplicate the logic on the client-side and server-side for anything that accesses the `session`, I have no problem with this but you [didn't like it before](https://github.com/kitspace/kitspace-v2/issues/129#issuecomment-853914428).

kasbah, 2021-09-20:
I think we need to do a pair programming session to figure this out.

PR #208 (2021-09-29): Server vs client rendering
Fixes #129
closed 2021-09-29 by kasbah