kitspace/kitspace-v2 #133: Error on staging: window.plausible is not a function

kasbah, 2021-06-10:
Can be seen on https://master.staging.kitspace.dev

```
TypeError: window.plausible is not a function
    NextJS 54
```

Strangely it's causing a "404" error, but also saying "An unexpected error occurred".

kasbah, 2021-06-10:
I'm just updating the extension to include `kitspace.dev` so it's probably this one firing:

https://github.com/kitspace/kitspace-v2/blob/6da83c53f72aa9efb17606d83b02ebb804647b78/frontend/src/components/Board/BuyParts/index.js#L59-L65

Though, that shouldn't be calling plausible till someone actually clicks on a button?

kasbah, 2021-07-18:
This was solved by #153, we should make sure it's not firing when just rendering the page though.

AbdulrhmnGhanem, 2021-09-12:
> we should make sure it's not firing when just rendering the page though.

Can you elaborate please?

kasbah, 2021-09-14:
If it was causing an error, just by loading the page, then somehow `window.plausible` was being invoked on load. We should doubly make sure these functions aren't being called when they shouldn't be.

closed 2021-10-30 by kasbah