kitspace/kitspace-v2 #134: Assets not found on staging

kasbah, 2021-06-10:
https://master.staging.kitspace.dev/kaspar/ulx3s

```
GEThttps://master.staging.kitspace.dev/static/images/aisler.png
[HTTP/1.1 404 Not Found 17ms]

GEThttps://master.staging.kitspace.dev/static/images/pcbway.png
[HTTP/1.1 404 Not Found 27ms]

GEThttps://master.staging.kitspace.dev/static/images/jlcpcb.png
[HTTP/1.1 404 Not Found 24ms]

GEThttps://master.staging.kitspace.dev/static/images/oshpark.png
[HTTP/1.1 404 Not Found 27ms]

GEThttps://master.staging.kitspace.dev/static/images/pcbshopper.png
[HTTP/1.1 404 Not Found 27ms]

GEThttps://master.staging.kitspace.dev/static/images/Digikey.ico
[HTTP/1.1 404 Not Found 27ms]

GEThttps://master.staging.kitspace.dev/static/images/Mouser.ico
[HTTP/1.1 404 Not Found 31ms]

GEThttps://master.staging.kitspace.dev/static/images/RS.ico
[HTTP/1.1 404 Not Found 41ms]

GEThttps://master.staging.kitspace.dev/static/images/Newark.ico
[HTTP/1.1 404 Not Found 40ms]

GEThttps://master.staging.kitspace.dev/static/images/LCSC.ico
```

AbdulrhmnGhanem, 2021-06-11:
Are there differences between the build config for this branch and abdo-dev.staging.kitspace.dev?

kasbah, 2021-06-18:
I believe this was a migration issue with the volumes. Removing all volumes got rid of it. We have to watch out for things like this though.

closed 2021-06-18 by kasbah