kitspace/kitspace-v2 #138: Integrate multi project in frontend

AbdulrhmnGhanem, 2021-06-11:
@kasbah, can you please give reference to some projects which use multi configuration to use for testing?

AbdulrhmnGhanem, 2021-06-17:
I think we have two options for the homepage
1. After getting the repos from gitea, make another request to the processor to find whether it's a multi-project or not. Then create cards for the subprojects.
2. Make a single card out of each repo. For multi-project repos, the `[username]/[projectName]` page will have a structure similar to the one on the homepage, a card for each subproject.

I don't like both, before thinking of the implementation, I thought the first option would be straightforward, it could be simple if the repo itself has a `project_type`: `single` | `multi` or something like that.

kasbah, 2021-06-17:
I prefer no 1 definitely. How to best go about it I am less certain on. If we need to further modify Gitea, so be it. I think just implement it with the extra requests to the processor for now, we'll see how it performs.

PR #147 (2021-07-23): Integrate multi project in frontend
closes #138
closed 2021-07-23 by kasbah