kitspace/kitspace-v2 #148: Add processor endpoint to indicate overall processing status

AbdulrhmnGhanem, 2021-07-12:
Instead of querying each file status individually, aggregate the `in_progress` status.

kasbah, 2021-07-12:
The processor architecture is kind simple in this regard. It's not aware of all the files that willl be created,it accepts and shows any files the processing function say they will create in the moment. This makes it easier to add new files and endpoints.

I'm not sure the overall processing status is that useful. No one page will want to show _all_ the files the processor creates. On the frontend you have a better idea of what files you want to display for a particular page. I think you should aggregate the progress based on the files you want to show on that page.

AbdulrhmnGhanem, 2021-07-12:
It will be used primarily to display a processor spinner when migrating a repo. It's pretty handy for testing without using `cy.wait`.

kasbah, 2021-07-12:
Well, is it a big thing to aggregate all the endpoints needed to show the initial page? The frontend will know better what's needed and can adapt appropriately. Also, handy for testing is the worst reason: the cypress tests really need to be independent of the backend API and should only look for DOM elements.

AbdulrhmnGhanem, 2021-07-12:
> Also, handy for testing is the worst reason: the cypress tests really need to be independent of the backend API and should only look for DOM elements.

> It will be used primarily to display a processor spinner when migrating a repo.

So cypress will wait until this spinner disappears.

kasbah, 2021-07-12:
Ah, sorry, I missed that. Yes, wait for the spinner in Cypress. The spinner is pretty handy for people too though :D 

Can you aggregate the status in the frontend based on the files you need for the project page though? Seems like the better way to go for me.