kitspace/kitspace-v2 #158: Staging server running out of space

AbdulrhmnGhanem, 2021-07-17:
The deployment fails because there is no space left on the device. https://github.com/kitspace/kitspace-v2/runs/3092839149?check_suite_focus=true#step:10:444

kasbah, 2021-07-17:
Well, log in and clear it out. :D

There should also be a cron job to remove docker images and volumes: https://github.com/kitspace/kitspace-v2/blob/master/ansible/roles/staging_server/tasks/docker.yml#L23

It might not be working correctly then...

kasbah, 2021-07-17:
Hmm, seems like it's cleared out already.

```
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        39G  6.9G   32G  18% /
```

AbdulrhmnGhanem, 2021-07-17:
> Well, log in and clear it out. :D

I did so.
But meant if there's something we can do about it to prevent this from happening again.

kasbah, 2021-07-17:
Right, yeah, that's what that cron job is for. I guess it's not often enough. Maybe we need to clear it out as part of the deploy.

kasbah, 2021-07-17:
I think https://github.com/kitspace/kitspace-v2/commit/da9720c9f7e1220ed96218c3ba32cc2ca5559e1b fixes this, feel free to re-open if it happens again.

commit da9720c9 (2021-07-17): Fix pruning of docker containers on staging servers

fixes #158
closed 2021-07-17 by kasbah