kitspace/kitspace-v2 #160: Long response time when clicking project cards

kasbah, 2021-07-18:
When clicking on a project on the main page in staging it takes a noticeable long time before anything happens. Even before the spinner shows (which we should get rid of for #131). Culprit seems to be the NextJS server serving up the required JSON. Probably this is due to too much happening in the `getServerSideProps`. 

[image]

AbdulrhmnGhanem, 2021-07-18:
Yes, I think we should only make two requests one to Gitea and one to the processor.

https://github.com/kitspace/kitspace-v2/pull/126#issuecomment-850997520

kasbah, 2021-07-18:
While improving the processor API to be a single endpoint would help I'd like to see how far we can get with separate endpoints. They do have an advantage in that the front-end can decide what it wants to display and only request that. 

I think this response time can be improved by using `Promise.all` similar to what you did on the mulri page:

https://github.com/kitspace/kitspace-v2/blob/efe50fedec3d2537999ab843c21c947cc0b8f7ee/frontend/src/pages/[username]/[projectName]/[multiProjectName]/index.js#L36-L52

kasbah, 2021-07-18:
Hmm, I guess what I looking at on abdo-dev.staging already was fetching all that concurrently. So it's not enough...

AbdulrhmnGhanem, 2021-07-21:
I think we should profile it, instead of guessing which part is the bottleneck.

AbdulrhmnGhanem, 2021-07-26:
Have you tried [clinicjs](https://clinicjs.org/) before?

kasbah, 2021-07-26:
I've not come across it. Seems a bit overkill for this specific case maybe?

AbdulrhmnGhanem, 2021-08-21:
I took a quick look. The CPU is mostly idle, most of the time is spent blocking on IO, probably the processor requests.
I will mock the processor requests and see if it solves the issue.

kasbah, 2021-09-01:
Related to #185

AbdulrhmnGhanem, 2021-09-09:
# Stats for ULX3S in the dev environment
* All requests to the processor resolve in `~250ms`.
[image]

##
I mocked the assets and gradually fetched the real assets, the summary below is the average of 4 samples.
| assets used| aggregate time for loading JSON|
|-----------------|------------------------|
|none| 100ms|
|`getKitspaceYAMLJson`| 183ms|
|`hasInteractiveBom` | 215ms|
| `getIsProcessingDone` | 232ms|
| `getBoardGerberInfo`| 252ms|
|`getBoardBomInfo` | 296ms|
| `getDefaultBranchFiles` | 755ms|

So the bottleneck is `getDefaultBranchFiles`, the files are needed for readme processing. I will finish #128 then test again.

AbdulrhmnGhanem, 2021-09-10:
For ULX3S on the staging server, the response time in #202 is `250ms`-`300ms` when navigating by clicking on the card, and `400ms`-`600ms` when visiting the URL directly, previously it was `1200ms`.

AbdulrhmnGhanem, 2021-09-10:
After activating caching (78f5c8412ca07c5cfccebf9b6f833726caf3b53f) in Nginx the improvement is negligible.
It gets "HIT" for all the processor assets though there's no reduction in the total loading time of the page, so I guess it's no longer the bottleneck.

kasbah, 2021-09-14:
Surely Nginx is will cache on the first request though, so the first request is still slow.

AbdulrhmnGhanem, 2021-09-14:
These stats are without caching, I haven't merged the caching yet.

kasbah, 2021-09-14:
Alright but you said "After activating caching (...) there's no reduction in the total loading time of the page", so I meant: it being cached on the first request can explain that. Maybe it's still worth activating the caching.

AbdulrhmnGhanem, 2021-09-14:
Sorry for the confusion!

kasbah, 2021-10-30:
Seems to have improved, I'll open new issues if I notice any performance problems.

closed 2021-10-30 by kasbah