kitspace/kitspace-v2 #185: Delegate serving processor static files to nginx

AbdulrhmnGhanem, 2021-08-10:
Instead of using `express.static` for serving processor-generated files, rely directly on Nginx.

kasbah, 2021-08-10:
What would this get us, slightly better performance that we probably won't even notice?

There's some dynamic stuff going on with figuring out if the relevant file is being processed or failed. If we decide we don't need the special http return codes then it might make sense to just let Nginx handle it. A more future-looking proposal would be to move the files to an s3-like object storage but I'm also reluctant to do that until we actually hit a scale where it makes sense.

kasbah, 2021-09-01:
If performance is a lot worse than I think then we can get rid of the custom http responses and use Nginx.

AbdulrhmnGhanem, 2021-09-12:
https://github.com/kitspace/kitspace-v2/issues/160#issuecomment-917025548

kasbah, 2021-09-14:
That comment doesn't show me how serving directly with Nginx wouldn't be beneficial. Caching is one thing but a fast server and proper compression is another. 

Can you link me the issue that made us think that Express.static is slow?

AbdulrhmnGhanem, 2021-09-14:
> Can you link me the issue that made us think that Express.static is slow?

https://github.com/expressjs/express/issues/321

> Caching is one thing but a fast server and proper compression is another.

What I understood from the documentation is the request won't hit the node server at all if the response is cached (it's cached in Nginx not in user browsers), and if gzip is enabled it will work with the cached response too.

Also, I meant it won't provide any speed up because processors assets are no longer the bottleneck, even if we made it faster it won't speed up the total loading time.

AbdulrhmnGhanem, 2023-02-16:
Static files are in S3 now :tada:

closed 2023-02-16 by AbdulrhmnGhanem