kitspace/kitspace-v2 #269: Mix of styled-jsx and scss modules

kasbah, 2022-02-02:
The original justification for adding `styled-jsx` (9a4c6c6f94cf637e09ad6bd92d673c1076192bf4) was that it's needed for `dangerouslySetInnerHTML` which we only use for the readme. Now it's being used all over the place and being mixed it with scss modules. We should pick one or the other and use that, i.e. one of:
- Use it just for the readme and scss modules everywhere else
- Use `styled-jsx` everywhere
  
Any preference @AbdulrhmnGhanem ?

(Also using `<style global` everywhere. What's the point of breaking out of the component scope all the time?)

AbdulrhmnGhanem, 2022-02-02:
`styled-jsx` is used for styling elements that aren't under our control for example the readme and the parts of semantic-ui which aren't customizable via `scss`.
`global` has been used for the same reason, these elements aren't under the component control for example

|w/ global|w/o global|
|---|---|
|[image]|[image]|

We can move the jsx styles into `_app.scss` but I prefer keeping the style in the parent component.

kasbah, 2022-02-02:
Ok, when I said "all over the place" I forgot to actually check. Seems like that isn't true. :rofl: It's used twice in `NavBar` though. 

The readme makes sense I think (maybe there's a better way?). The one I was looking at is this: it sets an id in the component and then a `<style global jsx` to zero back in on the id modify it? That doesn't make a whole lot of sense. Is that really necessary?
https://github.com/kitspace/kitspace-v2/blob/ae711eb45e986fbfe19d3d57e98c4d950913eae8/frontend/src/components/NavBar/index.js#L91-L106

AbdulrhmnGhanem, 2022-02-02:
Yes this one doesn't need `global`.

kasbah, 2022-02-02:
And could be set from `module.scss`?

AbdulrhmnGhanem, 2022-02-02:
> And could be set from module.scss?

Yes, c9a15a76df43d250bfb1af69f66e7705ebf0a0d7.

kasbah, 2022-11-06:
Since it confused me once again, let's stop using it in the menu and add a comment to the readme one.

PR #496 (2022-11-22): Remove styled-jsx
- Refactor: move SmallMenu style into its SCSS module
- Refactor: replace styled-jsx with scss file for items alignment
- Refactor: replace styled-jsx with scss file for readme
- Deps: remove styled-jsx
##
|before (menu)|after (menu)|
|---|---|
|[image]|[image]|

|before (readme)|after (readme)|
|---|---|
|[image]|[image]|

##
Closes #269.
closed 2022-11-22 by kasbah