kitspace/kitspace-v2 #272: Is the whole page being remounted on transitions?

kasbah, 2022-02-02:
I'm trying to figure out what's going on with the page transitions and how we can fix it. The experience I'd like to get back to is what you get with a regular next app. 

Here is [a demo repo](https://github.com/kasbah/next-app-test). You can clone that, run the dev server, and click between "go to page two" and "go to page one". That's the kind of page transition we want, what's happening with ours? We get layout shift, flash of a loader, and issues like #268. Is everything being dismounted and mounted again unnecessarily?

kasbah, 2022-02-02:
Alright, let me know if you run out of ideas and I'll continue my investigation. One thing I did do was run the [almost vanilla create-next-app behind the `/__kitspace` gitea hack to check if it's that](https://github.com/kitspace/kitspace-v2/tree/kaspar/debug-page-transitions/frontend) (it's not).

AbdulrhmnGhanem, 2022-02-02:
I am trying this [blog](https://adamwathan.me/2019/10/17/persistent-layout-patterns-in-nextjs/) suggestions.

kasbah, 2022-02-02:
Those blog suggestions seem to be complicating things even more. I feel like we had this and then lost it, hence we have it when go back to vanilla create-next-app. (Even if create-next-app is still doing more re-rendering than needed according to that blog.) I suspect it's something to do with the auth and the use of `useEffect` which we never fully straightened out.

kasbah, 2022-02-11:
I think most of this is solved with #280, there's still some "blinking" I can see on Firefox but I should probably make a more specific issue.

closed 2022-02-11 by kasbah