kitspace/kitspace-v2 #282: The CI has become so confusing

AbdulrhmnGhanem, 2022-02-03:
The name of the jobs isn't the name of the commits. Also, why does it keep merging unrelated branches into `review`?

I think the previous workflow was much better the only we needed is to merge a pr into `review` whenever it's marked as ready for review. And I don't what triggers a new build now, #276 is failing on a deleted test!

kasbah, 2022-02-03:
I changed two major things about the CI:

1. Using `pull_request_target` trigger in all workflows.
- I think  this is confusing you the most. This triggers the CI when pull-requests are made to the master branch. 
 - Unlike `pull_request` it has the same security privileges as `push` events (`push` runs on push to any branch -- what we were using before) but `pull_request_target` also runs the _workflow_ file that is on `master` _not the pull-request branch_.  (There's a longer [article from Github](https://securitylab.github.com/research/github-actions-preventing-pwn-requests/) on this that might help). 
- This does makes debugging/working on the workflow itself a bit harder. (One trick is to add the branch you are currently working on to the `on: push` array and remove it again when it's ready -- you still get failures from the `pull_request_target` runs but at least you can confirm it works)
- The reason I made this change is so that Dependabot pull-requests can run our CI without getting errors about missing secrets. 

2. The other aspect is the `review` branch merging. Yes, it merges unrelated branches into `review` and deploys them. But it's as you say: only branches that are marked ready for review. (So I'm not clear on what you want to change there). The effect of this is that you can basically ignore all the CI runs with the `review` tag and just rely on the "github-actions bot deployed to review.staging.kitspace.dev" statuses on the PRs. 

The two things are unrelated. So we can still change one without affecting the other, but it seems you are confused because I changed two major things in a short time span. I would say: give it some time and if you still find it annoying after changing your mental model about what's happening then we can look at changing the `pull_request_target` strategy but still avoid Dependabot PRs failing.

AbdulrhmnGhanem, 2022-02-03:
I don't understand this one: in 3a73e49 the `pull_request_target` is failing because it's running a deleted test! What may cause this?
 
[image]

kasbah, 2022-02-03:
I'm not sure. I assumed it would be because something was deleted from in docker.yml but it's running the docker.yml from `master` but that is not the case?

kasbah, 2022-02-03:
Sorry, I figured it out. This checkout:

https://github.com/kitspace/kitspace-v2/blob/a9d50f0b72c95c43077e8871dc4fb7ff0340c9be/.github/workflows/docker.yml#L232-L235

Needs to be changed to be like this checkout

https://github.com/kitspace/kitspace-v2/blob/a9d50f0b72c95c43077e8871dc4fb7ff0340c9be/.github/workflows/docker.yml#L86-L94

I needs to be changed on the master branch docker.yml!

AbdulrhmnGhanem, 2022-02-04:
I think the `review` shouldn't run the `e2e` especially after enabling running tests in parallel any run of the `e2e` will spin up 12 jobs  (GitHub Actions free plan supports 20 concurrent containers).

kasbah, 2022-02-04:
I've been thinking maybe we should stop testing across different browsers for the e2e tests. Seems like a lot of repeated tests that are not really about browser compatibility. Functional tests on a component level seem like a better target for cross-browser tests.

How come it's 12 now anyway? What's this new `containers` matrix about?

https://github.com/kitspace/kitspace-v2/blob/3123cb2a2c25e256d7474ec1b446de32f5ea5cd5/.github/workflows/docker.yml#L223-L225

AbdulrhmnGhanem, 2022-02-04:
> How come it's 12 now anyway? What's this new containers matrix about?

[Running the tests in parallel](https://docs.cypress.io/guides/continuous-integration/github-actions#Worker-Jobs).

kasbah, 2022-02-04:
But [1, 2, 3] are just repeating the same tests on the same browser?