kitspace/kitspace-v2 #310: README image are broken again

AbdulrhmnGhanem, 2022-02-14:
* Gitea changed how redirection works; it broke README URLs,
* Also, should add a check for this because I found the bug accidentally.

AbdulrhmnGhanem, 2022-02-14:
It's broken for projects with capital letters in its name, e.g., CH330_Hardware.

AbdulrhmnGhanem, 2022-02-15:
This URL works https://gitea.review.staging.kitspace.dev/tester2/CH330_Hardware/raw//images/back.png
But this doesn't https://gitea.review.staging.kitspace.dev/tester2/ch330_hardware/raw//images/back.png

The only difference is `CH330_Hardware` in the working URL and `ch330_hardware` in the broken one.

kasbah, 2022-02-15:
That actually seems like a Gitea bug to me. Since https://gitea.review.staging.kitspace.dev/tester2/ch330_hardware re-directs so should any sub-path of it. 

(We need to also fix does double slashes (`//`) we are adding in a few URLs by the way. I've noticed them before, may not even have been readme related.)

AbdulrhmnGhanem, 2022-02-15:
Specify the branch name fixes it: https://gitea.review.staging.kitspace.dev/tester2/ch330_hardware/raw/branch/master/images/back.png

We were relying on undocumented behavior so I don't think it's a real bug. Now we have to pass the branch name 

https://github.com/kitspace/gitea/blob/4d4a090cc98c8a67634d2b4dc0dbc0eac0523d88/routers/web/web.go#L1025

AbdulrhmnGhanem, 2022-02-15:
Opened a PR upstream https://github.com/go-gitea/gitea/pull/18775.

AbdulrhmnGhanem, 2022-04-12:
Closed via #311.

closed 2022-04-12 by AbdulrhmnGhanem