kitspace/kitspace-v2 #337: Wrong files found for ofm-illumination

kasbah, 2022-05-09:
(When uploading the folder)

[image]
[image]
https://gitlab.com/openflexure/openflexure-constant-current-illumination

AbdulrhmnGhanem, 2022-05-18:
When uploading, all files get committed to the repo root without any folder structure
|Uploaded|Synced|
|---|---|
|[image]|[image]|

AbdulrhmnGhanem, 2022-05-18:
This bug is reproducible since https://github.com/kitspace/kitspace-v2/commit/358d9c5335d185a560578ed7d3ef7ddc17bea32a,
|358d9c5|395df65 (one commit before)|
|---|---|
|[image]|[image]|

AbdulrhmnGhanem, 2022-05-18:
We had to upload via multiple commits because committing files is responsible for the `tree_path` yet it was annoyingly slow.
I think the proper solution is adding multi-file uploading support in gitea (upstream) itself. @kasbah, are you OK with this?