kitspace/kitspace-v2 #362: `aux-ps` wrong readme

AbdulrhmnGhanem, 2022-05-23:
|v1|v2|
|---|---|
|[image]|[image]|

AbdulrhmnGhanem, 2022-05-23:
This is a problem across all [github.com / eez-open](https://github.com/eez-open/modular-psu) multi-projects.

kasbah, 2022-05-23:
So initially we thought we would make the multi yaml key the "folder". This caused all sorts of annoying things though, including having to deal with slashes in the keys as you see on #359. Later we made people just pretend the yaml key is the name and make everything else explicit, including the `readme:`.  I.e.

```yaml
multi:
  project_name:
    readme: project/folder/readme.md
    gerbers: project/folder/gerbers/
    ...
```

We should probably change it to this, and then it finds it automatically:

```yaml
multi:
  project_name:
    folder: project/folder
``` 

Though if  the rest of the paths are then relative to that folder needs to be decided too. 

We should decide on something and then send PRs to the few projects that need to update their kitspace.yaml afterwards.

AbdulrhmnGhanem, 2023-02-16:
https://github.com/eez-open/modular-psu/pull/115

AbdulrhmnGhanem, 2023-02-16:
https://github.com/Jana-Marie/hoverboard-breakout/pull/12

AbdulrhmnGhanem, 2023-02-20:
Both are merged in the project's repos.

closed 2023-02-20 by AbdulrhmnGhanem