kitspace/kitspace-v2 #367: Limit memory of docker containers

kasbah, 2022-05-24:
As you said on chat @AbdulrhmnGhanem 

> I think we should limit the memory usage of the containers, if this happens again the container will crash instead of taking the whole machine down

AbdulrhmnGhanem, 2023-02-20:
I delete the imported projects and the files in the S3 bucket and imported everything again to see the usage during the import operation.
As expected  `gitea` and the `processor` are the major consumer of resources.

I think limiting everything to `70%` would be OK; none of the containers reach `70%` (mem or cpu)  so there is a bit of wiggle room and we still have `30%` for the host machine itself.

[image]

kasbah, 2023-02-20:
> I think limiting everything to 70% would be OK; 

That's 70% of the host machine?

AbdulrhmnGhanem, 2023-02-20:
yes

AbdulrhmnGhanem, 2023-07-24:
Here is the patch to do so :point_down: 

```diff
diff --git a/docker-compose.deploy.yml b/docker-compose.deploy.yml
index ed3a59a7..f0e619f1 100644
--- a/docker-compose.deploy.yml
+++ b/docker-compose.deploy.yml
@@ -1,4 +1,5 @@
-version: '2.3'
+version: '3.4'
+
 services:
   frontend:
     image: ghcr.io/kitspace/frontend${FRONTEND_DEPLOY_IMAGE_TAG}
diff --git a/docker-compose.e2e.yml b/docker-compose.e2e.yml
index a7e06329..2884be27 100644
--- a/docker-compose.e2e.yml
+++ b/docker-compose.e2e.yml
@@ -1,4 +1,4 @@
-version: '2.3'
+version: '3.4'
 
 services:
   frontend:
diff --git a/docker-compose.override.yml b/docker-compose.override.yml
index 3dc05955..79ce2d5a 100644
--- a/docker-compose.override.yml
+++ b/docker-compose.override.yml
@@ -1,4 +1,4 @@
-version: '2.3'
+version: '3.4'
 
 services:
   frontend:
diff --git a/docker-compose.test-processor.yml b/docker-compose.test-processor.yml
index 614feb8c..8f21bfbf 100644
--- a/docker-compose.test-processor.yml
+++ b/docker-compose.test-processor.yml
@@ -1,4 +1,4 @@
-version: '2.3'
+version: '3.4'
 
 services:
   processor:
diff --git a/docker-compose.yml b/docker-compose.yml
index 6f71459c..0a36824c 100644
--- a/docker-compose.yml
+++ b/docker-compose.yml
@@ -1,4 +1,4 @@
-version: '2.3'
+version: '3.4'
 
 services:
   frontend:
@@ -94,8 +94,11 @@ services:
     volumes:
       - gitea-data:/gitea-data:ro
       - processor-data:/data
-      # share docker daemon when running docker inside docker
-      - /var/run/docker.sock:/var/run/docker.sock
+    deploy:
+      resources:
+        limits:
+          cpus: '0.7'
+          memory: 22485M
 
   redis:
     image: redis:6.2-alpine
```
But we have the use `docker swarm` instead of `docker-compose`. 

https://github.com/kitspace/kitspace-v2/actions/runs/5649435082/job/15303783726#step:8:61
```logs
Command: docker-compose -f docker-compose.yml -f docker-compose.deploy.yml pull
Some services (processor) use the 'deploy' key, which will be ignored. Compose does not support 'deploy' configuration - use `docker stack deploy` to deploy to a swarm.
```
I am not sure whether it's worth the transition or not.

kasbah, 2023-07-24:
I know very little about swarm so it's probably not worth it right now. I think we could keep using version 2.3 and use the older mem_limit? https://stackoverflow.com/questions/56399794/how-to-limit-memory-usage-in-docker-compose

AbdulrhmnGhanem, 2023-07-25:
Tried it and found another problem, it doesn't seem to set it properly: 
https://stackoverflow.com/questions/74478539/docker-compose-ignores-memory-reservation-and-memory-limit