kitspace/kitspace-v2 #371: `nunchuk_breakout` wrong board showcase

AbdulrhmnGhanem, 2022-05-27:
https://kitspace.org/boards/github.com/jana-marie/hoverboard-breakout/nunchuk_breakout/

|v1|v2|
|---|---|
|[image]|[image]|

Though the IBOM is correct!
|v1|v2|
|---|---|
|[image]|[image]|

AbdulrhmnGhanem, 2022-07-28:
Both command
https://github.com/kitspace/kitspace-v2/blob/98af50fdd5b201fb368affd5c7f7c70edac32b62/processor/src/tasks/processKicadPCB/index.ts#L70
and 
https://github.com/kitspace/kitspace-v2/blob/98af50fdd5b201fb368affd5c7f7c70edac32b62/processor/src/tasks/processKicadPCB/index.ts#L79
fail, but the error is returned in `stderr` so move this error into node we have to use something like
```js
if (stderr) {
  throw new Error(stderr)
}
```
But this doesn't explain why the ibom is right while the kicad file is corrupt.

AbdulrhmnGhanem, 2022-07-28:
Actually, they fail for other boards too! It's an assertion not error
```
 failed to plot kicad gerbers
failed tester/ch330_hardware/059d467302fa712f2a8359c999bac46539a83ed7/images/layout.svg ../src/common/stdpbase.cpp(62): assert "traits" failed in Get(): create wxApp before calling this
../src/common/stdpbase.cpp(62): assert "traits" failed in Get(): create wxApp before calling this
../src/common/stdpbase.cpp(62): assert "traits" failed in Get(): create wxApp before calling this
```
Now I am not sure this is the root cause of this issue.

kasbah, 2022-07-29:
stdpbase.cpp is used by Kicad  It fails on these .kicad_pcb because they were last saved with a nightly build of Kicad.

kasbah, 2022-10-03:
It's using the folder provided in kitspace.yaml so it's not doing anything wrong as far as I can tell.

https://github.com/Jana-Marie/hoverboard-breakout/pull/11

PR #402 (2022-08-25): Handle sub project names with special characters
Fixes #371
closed 2022-10-03 by kasbah