kitspace/kitspace-v2 #374: `ArduTouch` missing project description

AbdulrhmnGhanem, 2022-05-27:
|v1|v2|
|---|---|
|[image]|[image]|

kasbah, 2022-05-27:
This is down to not support a feature of kitspace-v1 yet: retrieving the description from a Github repo through the Github API.

AbdulrhmnGhanem, 2022-05-28:
Actually, we fall back to the Gitea repo description but somehow it doesn't get imported!

[image]

kasbah, 2022-05-28:
Probably because we do a "Git" migration not a "Github" migration. Do we never want to migrate issues and wiki etc.

AbdulrhmnGhanem, 2022-05-29:
> Probably because we do a "Git" migration not a "Github" migration

You are right!

PR #393 (2022-06-01): Add service information when syncing repos
Fixes #374
closed 2022-06-01 by kasbah