kitspace/kitspace-v2 #394: Processing is flaky and so slow

AbdulrhmnGhanem, 2022-05-31:
Most of the migration fails (for projects that were working OK before) and we need a better solution for handling memory usage, throttling the process by reducing the concurrent jobs isn't ideal.

kasbah, 2022-06-01:
This is isn't a very actionable issue. Please provide more concrete examples of what you tried and what happened. Ideally keep an eye on the memory usage of the staging server (e.g. with `htop`) while you are doing it and let me know what processes exactly are taking up how much memory and for how long. 

Reducing concurrency or increasing available memory (running the processor across multiple servers if we have to) are the only ways to address this as these tasks simply need the resources they need.

kasbah, 2025-02-22:
Closing as a lot has changed since then and this issue is not actionable.

closed 2025-02-22 by kasbah