kitspace/kitspace-v2 #408: Show processor logs in CI if the `test_processor` workflow fails

AbdulrhmnGhanem, 2022-08-30:
There is no way to know what actually happened without running the tests locally.

kasbah, 2022-08-30:
Huh? There's a mocha test output and other logging from the runs e.g. https://github.com/kitspace/kitspace-v2/runs/8089438085?check_suite_focus=true#step:4:1858

AbdulrhmnGhanem, 2022-08-30:
I meant the output of the container itself, `docker-compose log processor or `docker-compose up processor`. The output of the test running sometimes isn't sufficient.

AbdulrhmnGhanem, 2022-09-02:
Actually, the output of running the tests locally is the same as CI. Even running `docker logs <test-container-name>` show the test result. We only get the same output as `docker-compose up` if we run `yarn start` and then `yarn test`.