kitspace/kitspace-v2 #450: Review processor is restarting because of git credentials

kasbah, 2022-10-13:
```
fatal: unable to auto-detect email address (got 'root@cfd641503233.(none)')

    at ChildProcess.exithandler (node:child_process:400:12)
    at ChildProcess.emit (node:events:513:28)
    at maybeClose (node:internal/child_process:1093:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:302:5)
```

Can you take a look please @AbdulrhmnGhanem. You can ssh in via deploy@review.staging.kitspace.dev

kasbah, 2022-10-13:
Looks like it was resolved but did you figure out why it happened?

AbdulrhmnGhanem, 2022-10-13:
I believe it's a side effect of something else; when I run `git config --list` in a working container there's no username or email both globally and per repo.
Though I couldn't get a shell into the container as it keeps crashing.

kasbah, 2025-02-18:
Old issue, closing.