kitspace/kitspace-v2 #480: Rate limiting by github fails silently

kasbah, 2022-11-03:
Only noticed this in gitea logs when doing migration e2e tests:

```
Run task failed: GET https://api.github.com/repos/kitspace-test-repos/DIY_particle_detector: 403 API rate limit of 60 still exceeded until 2022-11-03 13:23:52 +0000 GMT, not making remote request. [rate reset in 8m39s]
```

Not sure why Gitea using the API in the first place (to fetch the description?) but it's making the whole migration fail. We should avoid the failure but also report such migration failures in a better way.

kasbah, 2022-11-06:
Can happen in CI too:

https://github.com/kitspace/kitspace-v2/actions/runs/3405591812/jobs/5663695979#step:9:958

AbdulrhmnGhanem, 2023-07-03:
Moving this out of alpha-2; we are getting all repos' descriptions in a single request via the graphql API for now 
https://github.com/kitspace/kitspace-v2/blob/068dec4b0c3d70bf65bdf4ce5f42a042d023176b/scripts/importBoardsTxt.ts#L161-L206