kitspace/kitspace-v2 #491: frontend URLs forward compatibility

AbdulrhmnGhanem, 2022-11-07:
Make sure that URLs will still work after we support versioning via hashes.

Maybe we should use the same semantics as GitHub URL `{repo}/{hash}/{projectName}`

kasbah, 2022-11-09:
Definitely worth thinking about. The only issue with always having the hash/ref in the url is that it's quite ugly. We could continue with the scheme we have and then when we have git versions do something like:

- `{repo}/-/history/{hash}/{projectName}` 
- `{repo}/-/history/-/{projectName}` for the latest
- keep `{repo}/{projectName}` as an alias for the latest

For a single project it would look like:

- `{repo}/-/history/{hash}/_` 
- `{repo}/-/history/-/_` for the latest
- keep `{repo}` as an alias for the latest

kasbah, 2022-11-21:
Unless we can find some fault with the suggested URL scheme let's close this issue.

closed 2022-11-21 by kasbah