kitspace/kitspace-v2 #507: `tt_nano_HAT_b1` images getting CORS error

AbdulrhmnGhanem, 2022-11-24:
https://review.staging.kitspace.dev/tester/test_tube_photometer/tt_nano_HAT_b1

[image]

[image]

AbdulrhmnGhanem, 2022-11-24:
Actually, it isn't `tt_nano_HAT_b1` only, other projects too! But I don't know why we aren't getting this error for all projects!

kasbah, 2022-11-24:
This isn't happening for me.

AbdulrhmnGhanem, 2022-11-25:
This isn't happening anymore I don't know what was going on!

AbdulrhmnGhanem, 2022-11-25:
This is happening again even in firefox!

AbdulrhmnGhanem, 2023-01-11:
I am seeing 403 responses too now

[image]

[image]

[image]

AbdulrhmnGhanem, 2023-01-13:
Sending a request with the same browser headers using curl returns a response without the `Access-Control-Allow-Origin` header
```console
$ curl -i 'https://kitspace-staging-abdo-dev.s3.amazonaws.com/files/tester/DIY_particle_detector/HEAD/alpha-spectrometer/images/top.svg' -H 'User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:108.0) Gecko/20100101 Firefox/108.0' -H 'Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,*/*;q=0.8' -H 'Accept-Language: en-US,en;q=0.5' -H 'Accept-Encoding: gzip, deflate, br' -H 'Connection: keep-alive' -H 'Upgrade-Insecure-Requests: 1' -H 'Sec-Fetch-Dest: document' -H 'Sec-Fetch-Mode: navigate' -H 'Sec-Fetch-Site: cross-site' -H 'Pragma: no-cache' -H 'Cache-Control: no-cache'

HTTP/1.1 200 OK
x-amz-id-2: iVTdnGC9iImpV99C37aaJ1HM8nY03LVFp9ygo5K80VxlJg17Qg/ba2eXYAB2EsSmgLIExOcx5kk=
x-amz-request-id: 8TV0YT87XX586N14
Date: Fri, 13 Jan 2023 08:03:17 GMT
Last-Modified: Fri, 13 Jan 2023 07:35:33 GMT
ETag: "db35a5a9b9ed314d088c5d8db4b5ecd9"
Content-Encoding: gzip
Accept-Ranges: bytes
Content-Type: image/svg+xml
Server: AmazonS3
Content-Length: 49572
```

But sending a request with the origin and user agent only have the required headers
```console
$ curl -i -H "Origin: https:abdo-dev.staging.kitspace.dev" -H 'User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:108.0) Gecko/20100101 Firefox/108.0' https://kitspace-staging-abdo-dev.s3.amazonaws.com/files/tester/DIY_particle_detector/HEAD/alpha-spectrometer/images/top.svg

HTTP/1.1 200 OK
x-amz-id-2: RnqNly+1hfvlZ77DAOZ74X6KW5pDMckN1OcCFzeZXLqBLSoJFU/C2LHmhLA19/q13tegE/RhGhQ=
x-amz-request-id: H2E018Z24W910G2M
Date: Fri, 13 Jan 2023 08:06:40 GMT
Access-Control-Allow-Origin: *
Access-Control-Allow-Methods: GET, HEAD
Vary: Origin, Access-Control-Request-Headers, Access-Control-Request-Method
Last-Modified: Fri, 13 Jan 2023 07:35:33 GMT
ETag: "db35a5a9b9ed314d088c5d8db4b5ecd9"
Content-Encoding: gzip
Accept-Ranges: bytes
Content-Type: image/svg+xml
Server: AmazonS3
Content-Length: 49572
```

AbdulrhmnGhanem, 2023-01-13:
Disabling caching for this specific request fixes the issue.

[image]

I think head request shouldn't be cached in the first place because we are using it as a check.

PR #551 (2023-02-08): Don't cache `useProjectAsset` head request
Fixes #507.
It won't cause any performance issues; the body size is zero.
closed 2023-02-08 by kasbah