kitspace/kitspace-v2 #540: Gitea get served at the apex domain

AbdulrhmnGhanem, 2023-01-04:
https://master.staging.kitspace.dev/

This happened several times, so I don't think this is a random thing.

AbdulrhmnGhanem, 2023-01-05:
The `gitea` subdomain serves the frontend

[image]

AbdulrhmnGhanem, 2023-01-05:
I couldn't find anything in the logs. But restarting the nginx container fixed the issues.
So far we have seen several issues that gets solved by the restarting nginx. So I will add healthcheck that checks the web services (gitea, frontend, meilisearch) if restart nginx if one of them is inaccissible.

kasbah, 2023-01-05:
One thing that might be related is that Gitea and frontend are both on port 3000 on their respective containers. Maybe we can work around this by using a different port for the frontend?

AbdulrhmnGhanem, 2023-01-05:
If this is the case then there's something wrong with the docker network layer; we use the service name. https://github.com/kitspace/kitspace-v2/blob/342048d43f5a133d3afdea836798e95a62c2732f/nginx/kitspace.template#L55

I will change the port anyways, just in case.

kasbah, 2023-07-04:
Working around it with healthcheck restarting Nginx for now.