kitspace/kitspace-v2 #548: e2e failing on master, cypress not starting: ` error: unknown option: ---`

kasbah, 2023-01-09:
e.g. 

https://github.com/kitspace/kitspace-v2/actions/runs/3871738947/jobs/6599820302#step:9:488

AbdulrhmnGhanem, 2023-01-11:
The commit message needs some escaping
```
Bump json5 from 1.0.1 to 1.0.2 in /e2e (#546)
  
  Bumps [json5](https://github.com/json5/json5) from 1.0.1 to 1.0.2.
  - [Release notes](https://github.com/json5/json5/releases)
  - [Changelog](https://github.com/json5/json5/blob/main/CHANGELOG.md)
  - [Commits](https://github.com/json5/json5/compare/v1.0.1...v1.0.2)
  
  --- <---------------------------------------------
  updated-dependencies:
  - dependency-name: json5
    dependency-type: indirect
  ... <---------------------------------------------
  
  Signed-off-by: dependabot[bot] <support@github.com>
  
  Signed-off-by: dependabot[bot] <support@github.com>
  Co-authored-by: dependabot[bot] <[49](https://github.com/kitspace/kitspace-v2/actions/runs/3871738947/jobs/6599820302#step:9:49)699333+dependabot[bot]@users.noreply.github.com>
```

kasbah, 2023-02-08:
Would be good to get a fix in for this ASAP, it's kind of annoying.

AbdulrhmnGhanem, 2023-02-08:
OK, I got this!

PR #558 (2023-02-10): Handle special characters e2e commit info
I found escaping the commit body may be unnecessarily complicated and we don't need it in the cypress dashboard. So I removed it.
I rebased the depedabot commit on top of this commit to test that [it works](https://github.com/kitspace/kitspace-v2/actions/runs/4140118757).
##
Fixes #548
closed 2023-02-10 by kasbah