kitspace/kitspace-v2 #569: S3 assets versioning wrt processor version

AbdulrhmnGhanem, 2023-04-05:
Currently, if we make a change in the processor, to reprocess the projects we have to options:
1. Delete the S3 bucket, delete the data volumes, and re-import the projects again.
2. Add a new commit to all repos (probably out of our control).

I think we need a way to support versioning the assets with respect to the processor version. Maybe add the processor version to the file paths in S3?
This will be useful for rollbacks.

kasbah, 2023-04-07:
I've just read up on the S3 versioning a bit. I don't think it's  a solution on it's own because it just assigns a random id to each version of a particular object so there's no way to check "has this already been written by this new processor version?". 

One thing we could implement is that the processor overwrites all files on initial startup if say the env `KITSPACE_PROCESSOR_REPROCESS=true`.  We can still use S3 versions for rollbacks if needed. We also have the cache to contend with, set to 24hrs in Nginx I believe? (Is Next also caching assets actually?) 

I don't like adding the version to the path because then we keep having to update that in the frontend too and I don't see any benefit to having that control there.

AbdulrhmnGhanem, 2023-07-10:
Closed via #586.

closed 2023-07-10 by AbdulrhmnGhanem