kitspace/kitspace-v2 #625: Gitea build is failing

AbdulrhmnGhanem, 2024-01-06:
Run docker build for our fork without changing anything fails!

kasbah, 2024-01-07:
Are you getting this error too?

```
618.6 modules/hcaptcha/hcaptcha.go:11:2: unrecognized import path "go.jolheiser.com/hcaptcha": parse https://go.jolheiser.com/hcaptcha?go-get=1: no go-import meta tags (meta tag go.jolheiser.com/hcaptcha?go-get=1 did not match import path go.jolheiser.com/hcaptcha)
618.6 modules/password/pwn.go:11:2: unrecognized import path "go.jolheiser.com/pwn": parse https://go.jolheiser.com/pwn?go-get=1: no go-import meta tags (meta tag go.jolheiser.com/pwn?go-get=1 did not match import path go.jolheiser.com/pwn)
618.7 make: *** [Makefile:751: gitea] Error 1
------
Dockerfile:23
--------------------
  22 |     #Checkout version if set
  23 | >>> RUN if [ -n "${GITEA_VERSION}" ]; then git checkout "${GITEA_VERSION}"; fi \
  24 | >>>   && make clean-all build
  25 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c if [ -n \"${GITEA_VERSION}\" ]; then git checkout \"${GITEA_VERSION}\"; fi   && make clean-all build" did not complete successfully: exit code: 2
```

Best strategy might be to try and update Gitea and re-apply our patches.

AbdulrhmnGhanem, 2024-01-07:
> Are you getting this error too?

yes

kasbah, 2025-02-18:
We are not using a custom Gitea at the moment.

closed 2025-02-18 by kasbah